raspberrypi/pico-playground #4: Screenshots for video?

drudru, 2021-01-28:
It would be great to have some screenshots for the video demos. Other demos might benefit as well.

kilograham, 2021-02-21:
good idea

lurch, 2021-02-21:
As long as a screenshot of `sprite_demo` doesn't scare the children :wink:

BambooEngineer, 2021-04-01:
[image]

This is the color pattern i get, i dont think its my Resistor DACs but i could be wrong. The raspberry demo1 example looks fine for me but the floating heads has a black and blue box moving in the middle of the screen !

kilograham, 2021-04-01:
you probably forgot that there is a 1 pin gap between red and green

kilograham, 2021-04-01:
also i added some screenshots

closed 2021-04-01 by kilograham

BambooEngineer, 2021-04-01:
THANK YOU ! It worked