raspberrypi/pico-playground #8: 8080 and 6800 parallel bus example

siriokds, 2021-02-11:
On Rp2040 Datasheet page 324,
I can read: "The programmable input/output block (PIO) is a versatile hardware interface. It can support a variety of IO standards, including:  8080 and 6800 parallel bus..."

Is there an example about 8080 parallel bus ?
Tnx

lurch, 2021-02-11:
> Is there an example about 8080 parallel bus ?

Not yet, but we might add one when we're less busy with more-critical stuff!

This issue is closely related to https://github.com/raspberrypi/pico-examples/issues/48 ?

siriokds, 2021-02-12:
Yes. Thank you for your quick answer.

closed 2021-02-12 by siriokds

3s1d, 2022-09-21:
Hi,

did anything change in the past 1.5 years? I am about to start development on my 8080 based display and an example would really help me a lot (especially as this is my first time pio'ing something).

Thanks.

fu709691151, 2023-11-12:
Do you have any examples?