raspberrypi/pico-playground #9: Unable to run demo's on Pico Demo Board

JamesPeters98, 2021-02-16:
Hi! I've just got my VGA pico demo board and have tried running the examples on it, however I can't seem to get any VGA output at all, I have tested the pins for Hsync, Vsync and the colours and they seem to be outputting at around 31/16KHz or a multiple of them which seems correct to me.

So I'm wondering if my modern LCD monitor doesn't like these super-low resolution modes? My monitor detects a signal but just stays black.

Strangely, I also can't get audio to play, the USB sound card is the only example I could partially get running with the PWM outputting a very distorted sound in only the left channel.

I can't tell if I'm doing something wrong or my board is faulty.

kilograham, 2021-02-16:
does the Pico Demo come with any instructions; you can try passing something like -DPICO_BOARD=vgaboard to your cmake (do this from a fresh build directory)... the default pins for PICO_BOARD=pico are going to be wrong.

Note USB sound card currently only uses I2S so you must be listening to the raw I2S data!

kilograham, 2021-02-16:
that is the audio actually, for the video, make sure your examples/playground are up to date, there were changes in the last day or so to switch the 640x480 demos pixel clock from 24Mhz to 25Mhz by default. Most LCDs (other than TVs) don't seem to mind too much, but some may.

JamesPeters98, 2021-02-16:
Thanks for the quick reply! I'm using this board, https://shop.pimoroni.com/products/pimoroni-pico-vga-demo-base. And I cloned the playground, sdk and extras today from here, so that should be up to date.

Where should I be adding in the "-DPICO_BOARD=vgaboard" in my cmake? :)

JamesPeters98, 2021-02-16:
Oh nice, yeah I managed to change the PICO_BOARD variable to `vgaboard` and now it works perfectly, sweet! 

Probably would be an idea to add that into the CMake files or document it in the repo!

evolutional, 2021-02-17:
Had the same issue, this fixed it for me - thanks!

lurch, 2021-02-17:
> Probably would be an idea to add that into the CMake files or document it in the repo!

See Appendix D of https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf

closed 2021-02-18 by kilograham