raspberrypi/pico-playground #10: Popcorn Convertor and/or Samples

hoglet67, 2021-02-18:
Is there any possibility you could publish the video conversion tool and/or some samples for the Popcorn Media Player app in pico-playground.

This is what README.mb currently says:

[popcorn](apps/popcorn) This is a movie player for 30fps 320x240 movies with 44100 stereo sound, read in a custom format from SD card... it can even play backwards :-) TODO: add a link to big buck bunny in the right format TODO: add converter

Many thanks

kilograham, 2021-02-18:
ah yes... i'll upload the bunny movie today, and throw together some sort of instructions for the converter soon.

kilograham, 2021-02-18:
movie uploaded https://drive.google.com/file/d/1q3szTVccPZ08v_TMDxy9ZgqeOOXXwHCX/view?usp=sharing (also in the README.md which has some very basic instructions)

hoglet67, 2021-02-19:
Thanks for the Bunny example Graham.

It works, but the audio is very broken/stutterly.

Is that expected?

kilograham, 2021-02-19:
not expected; let me check the latest popcorn binary. (i've never seen audio problems with bunny, but sometimes i have with another movie - in which case pausing/unpausing a couple of times usually fixes it).

kilograham, 2021-02-21:
sound is good for me; did you get the sound working? I've mostly been using Samsung EVO Selects; though I have used a Sandisk in the past. (can't find it right now).

Certainly my SD testing ^ is not exhaustive, but we're not really pushing them very hard. Definitely worth trying with a different type of card though

kilograham, 2021-02-21:
oh, having just done this myself, be careful you didn't slow down the video by pressing the left button... the sound will stutter here. (this again is fixed by pause/unpause to set back to normal speed)

hoglet67, 2021-02-21:
> sound is good for me; did you get the sound working? I've mostly been using Samsung EVO Selects; though I have used a Sandisk in the past. (can't find it right now).

I've just tried again, and yes pausing/unpausing does resolve the sound issue.

I didn't notice this before, because I had a second source of similar (clicky) noise which I think was down to a USB ground loop

kilograham, 2021-02-21:
> BTW, it does start stuttering again each time you reset. So this must be down to the latency of the first read from the SD Card.

yup... `popcorn` is by no means fool proof at this point :-)

kilograham, 2021-02-27:
added converter (non trivial and requires Linux likely)

closed 2021-02-27 by kilograham