raspberrypi/pico-playground #11: sprite_demo fails on VGA board

8bitkick, 2021-02-24:
Built with `PICO_BOARD=vgaboard` the `sprite_demo` fails immediately at runtime with the following message:

```*** PANIC ***
System clock (192000000) must be an integer multiple of the requested pixel clock (25000000).
```

Changing the sys clock value to a multiple of 25MHz succeeds. Seemed to need 250MHz to render the whole screen but the OC is unstable after a minute or so on my Pimoroni VGA demo board:

```
set_sys_clock_khz(250000, true);
```

This may well not be the right fix :)

lurch, 2021-02-24:
What if you set a sys clock of 125MHz? (I'm just guessing as I've not got a VGA board myself!)
That was the speed that the RP2040 chips _were_ going to run at by default, and it was only later that they got bumped to 133MHz.

8bitkick, 2021-02-24:
Could well be right there this might have just been missed during this 125MHz update https://github.com/raspberrypi/pico-playground/commit/ef3bdfab11801d390f05e0cf52dd57fcab8b22cc

8bitkick, 2021-02-24:
...setting sys clock to 125MHz only renders a small slice of the screen for me. I'm sure there are some magic settings I am missing

8bitkick, 2021-02-24:
Nah thats just renders a small slice of the screen. I guess there are some magic settings I'm missing here

kilograham, 2021-02-24:
set PICO_SCANVIDEO_48MHZ=1 in the cmakelists.txt

kilograham, 2021-02-27:
fixe

closed 2021-02-27 by kilograham