raspberrypi/pico-playground #15: Licence for the Ubuntu Mono font

thejpster, 2021-03-28:
I _think_ that the font files:

* [pico-playground/scanvideo/textmode/font10.c](https://github.com/raspberrypi/pico-playground/blob/master/scanvideo/textmode/font10.c)
* [pico-playground/scanvideo/textmode/font8.c](https://github.com/raspberrypi/pico-playground/blob/master/scanvideo/textmode/font8.c)
* [pico-playground/scanvideo/textmode/font6.c](https://github.com/raspberrypi/pico-playground/blob/master/scanvideo/textmode/font6.c)

include a rendered copy of the font "Ubuntu Mono". If so, the licence (https://ubuntu.com/legal/font-licence) for that font states:

> Each copy of the Font Software must contain the above copyright notice and this licence. These can be included either as stand-alone text files, human-readable headers or in the appropriate machine-readable metadata fields within text or binary files as long as those fields can be easily viewed by the user.

I Am Not A Lawyer, and maybe rendering a font down to a bitmap doesn't constitute a "copy" of the font, but giving a link to Ubuntu as the source seems reasonable nonetheless. Some notes on how you rendered the font would also be handy, in case anyone wants a different font size.

lurch, 2021-03-28:
> Some notes on how you rendered the font would also be handy, in case anyone wants a different font size.

Based on the identifier names, it seems to be based on the same structure as https://docs.lvgl.io/latest/en/html/overview/font.html ?

kilograham, 2021-04-01:
added license

closed 2021-04-01 by kilograham