raspberrypi/pico-playground #16: Sprite_Demo

BambooEngineer, 2021-04-02:
I know the floating head sprites are arrays of char hex but i cant find an image converter that seems to work. When i replace the original sprite hex with something i get from any converters i do find, nothing appears. Not even a few pixels ! Im not a great coder but i wanna display my own sprites atleast. I have been successful with changing everything but the sprite itself like speed, position, bg color

kilograham, 2021-04-02:
@Wren6991 what did u use to create the data?

BambooEngineer, 2021-04-02:
LVGL.io and image2cpp, i also used arduboy image converter but im not surprised that didnt work cause thats a black and white conversion only since the arduboy is on a tiny lcd.

Wren6991, 2021-04-06:
With the `packtiles` script, also in this repo:

https://github.com/raspberrypi/pico-playground/blob/master/scanvideo/flash_stream/img/packtiles

E.g. `packtiles -sdmf bgar5515 raspberry.png raspberry_bgar5515.h` will convert a single image into the format expected by that sprite demo:

- Single image (`-s`)
- Dithered for better colour reproduction at low bit depth (`-d`)
- With opacity metadata included for faster alpha blitting (`-m`)
- To format BGAR5515, 3x5 colour bits and one alpha bit in BGAR order from MSB to LSB (`-f bgar5515`)

Wren6991, 2021-04-06:
You'll need python3 and PIL installed to run the script

BambooEngineer, 2021-04-11:
Nice i will try it out, 8 bit kick made a bgar5515 converter that outputs on the java console but only a few lines of my image showed up. Like the image was a face and all i got on screen where the eyes.

BambooEngineer, 2021-05-16:
the packtiles script launched and gave me data but it just did not work out on screen. I finally found a script that works created by boochow. Now i can use the example code from demo1 to do sprite movement.