raspberrypi/pico-playground #17: Problem with UART in hello_sleep example

ms1963, 2021-05-01:
I am currently experimenting with the sleep/dormant modes of the Raspberry Pi Pico. As a base I am using the examples hello_sleep respectively hello_dormant. The problem I encountered: After going to sleep, no printf() statements will be printed to the uart. 

Ther only text I'll ever see in the serial monitor is:
[Connected]
Hello Sleep!
Switching to XOSC

I am using a Mac computer. Thus, I don't know whether Windows 10 and Linux have the same issue.
Is there anything I need to consider when sleeping or recovering from sleep? I have tried  to save the clock registers before sleep and restoring them after sleep, but without any success.

RayNieport, 2022-08-12:
I have the same issue while monitoring stdio via USB on Windows 10. Dormant mode appears to shut down things without properly re-enabling them on wake-up.
I implemented the `recover_from_sleep` function found [here](https://github.com/ghubcoder/PicoSleepDemo/blob/master/main.c), but while that does fix some issues and appears to bring the clocks back up, it does not fix the USB and allow stdio to display on my monitor via PuTTY.

lurch, 2022-08-18:
This is just a random guess (and may not actually be anything to do with your problem at all), but I wonder if Windows might be "getting confused" by the USB device suddenly becoming non-responsive, when the RP2040 goes to sleep?

RayNieport, 2022-08-18:
Windows does seem to get confused, but I think this is more of a Pico issue than a Windows issue. I would think that a proper "wake up" after going dormant should reinitialize USB in such a way that Windows could recognize the device again. And as mentioned above, this also seems to be an issue on Mac. I've monitored the system clocks and by default the USB clock stays at 0Khz after waking up from dormant.

I tried to restore this manually but setting the clock back to its original value doesn't seem to be enough to get the connection working again.