raspberrypi/pico-playground #20: Incorrectly connected resistors in 'Raspberry Pi Pico to VGA Connection Schematic.png'

CHiPs44, 2021-11-14:
Hello,

It seems some resistors on 'Raspberry Pi Pico to VGA Connection Schematic.png' to GND pins instead of GPxx pins, and some do not go to the right pin.

For example, on the green output, 2k resistor should go to GP8/11 and 1k one should go to GP9/12.

lurch, 2021-11-15:
I've just cross-referenced https://github.com/raspberrypi/pico-playground/blob/master/scanvideo/Raspberry%20Pi%20Pico%20to%20VGA%20Connection%20Schematic.png (which was contributed by a 3rd-party in #19 ) against Figure 24 of https://datasheets.raspberrypi.com/rp2040/hardware-design-with-rp2040.pdf (which should be taken as definitive) and it appears you're correct. Looks like the HSYNC and VSYNC have been switched as well?

@ddonofrio Could you supply a corrected version of your diagram please?

TomKeddie, 2022-09-21:
@aallan would also be great to link to the kicad sources for the board in https://datasheets.raspberrypi.com/rp2040/hardware-design-with-rp2040.pdf :+1:

lurch, 2022-09-22:
> would also be great to link to the kicad sources for the board in https://datasheets.raspberrypi.com/rp2040/hardware-design-with-rp2040.pdf

Page 14 of that PDF says "Schematics and layout files are available for KiCad at https://datasheets.raspberrypi.com/rp2040/VGA-KiCAD.zip and https://datasheets.raspberrypi.com/rp2040/VGA-PicoW-KiCAD.zip. KiCad is a free, open-source suite of tools for
designing PCBs and can be found at https://kicad.org/."

lurch, 2023-05-03:
#41 seems to fix this.

kilograham, 2023-05-03:
fix merged

closed 2023-05-03 by kilograham