raspberrypi/pico-playground #22: USB Sound Card ceases function after some time of use

Envious-Data, 2022-01-24:
I have a couple of projects I'd like to use the USB Sound Card app for, however I Don't know enough about coding to fix the issue it has.

After aproximately 20 minutes it will stop working, the host usb port will also stop working and will not work again unless you reboot.

Is it possible that this issue could be fixed?

JamesH65, 2022-01-24:
What USB sound card are you having problems with, and on which Raspberry Pi device?

Envious-Data, 2022-01-24:
Im refering to:
https://github.com/raspberrypi/pico-playground/tree/master/apps/usb_sound_card
using a Raspberry Pi Pico

After some time of use it will stop working, my computer will play a notif
for a device being disconnected and that usb port my pico was plugged into
is unsuable till I reboot my computer.

lurch, 2022-01-24:
ping @liamfraser 
Note however that the code in this pico-playground repo is "officially unsupported", so your problem may not get fixed. There's some (different) USB audio examples in https://github.com/raspberrypi/pico-examples and that repo *is* "officially supported", so it might work better? :shrug: 

>  the host usb port will also stop working and will not work again unless you reboot.

_Maybe_ that indicates a problem with the USB drivers on your host, rather than with Pico?