raspberrypi/pico-playground #25: convert images for sprite_demo

TheLegendOfKitty, 2022-05-27:
I've tried multiple combinations with different bgar5515 converters (including the packtiles script) but all of them don't render properly (draws only 1 line, some lines are skipped). I'll provide images if anyone wants them. What is the proper way to convert images?

lurch, 2022-05-27:
Duplicate of #16 ?

TheLegendOfKitty, 2022-05-28:
Seems like OP fixed it, but didn't provide much detail on how. I tried the script he mentioned and did not have any luck.

zachlindsey, 2022-10-02:
Just in case anyone else finds this, here are some tips if you want to use the packfiles script:

1. Make sure your image is the right size. For sprite_demo, make it 128x128 OR edit the `berry[i].log_size = 7` line to be whatever power of 2 you want your images to be.
2. If your image has transparency, use `python packtiles.py -sdmf bgar5515 INPUT.png OUTPUT.h`. Note the `.h` on the output. The flags are important: `s` makes a single image, and `m` adds transparency metadata. If your image does NOT have transparency, remove the `m` tag.
3. If your image does NOT have transparency, you need to change `berry[i].has_opacity_metadata = false;`. Otherwise, the code will read part of your image as a mask. :)
4. Edit the produced `.h` file to have the name you want and edit the imports in the main file.

Good luck!