raspberrypi/pico-playground #30: How to use vga_mode_tft_800x480_50 mode?

barbiani, 2022-11-10:
Hello everybody,

I am experimenting with the flash stream example and a widescreen 800x480 tft.

Currently, with the project set to 640x480, the images are not completely shown as its corners are cut out.

The tft controller OSD shows 640x480 correctly. The PNG has the same format, but it is shown bigger than the screen.

I'd like to try higher resolutions as this txt supports 800x480.

setting VGA_MODE to vga_mode_tft_800x480_50 breaks with:

```
[ 44%] Linking CXX executable flash_stream.elf
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/flash_stream.dir/flash_stream.c.obj: in function `render_loop':
flash_stream.c:(.time_critical.render_loop+0x8c): undefined reference to `vga_mode_tft_800x480_50'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/flash_stream.dir/flash_stream.c.obj: in function `vga_main':
flash_stream.c:(.text.vga_main+0x14): undefined reference to `vga_mode_tft_800x480_50'
collect2: error: ld returned 1 exit status
make[2]: *** [scanvideo/flash_stream/CMakeFiles/flash_stream.dir/build.make:747: scanvideo/flash_stream/flash_stream.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:3058: scanvideo/flash_stream/CMakeFiles/flash_stream.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

It looks like pico_scanvideo_dpi is being used and is not including the proper vga configurations.

What can I do?

kilograham, 2022-11-10:
Yeah many of the mode timings are only specified when `PICO_SCANVIDEO_48MHZ` is defined to 1 (i.e. if you commit to running the system clock as a multiple of 48Mhz)

You are free to make your own mode for 125Mhz, but you'll have to figure out the timings yourself (unless someone wants to PR them)

CHiPs44, 2022-11-11:
Hello,

I tried to make a suitable mode with no success on my VGA demo board connected to an old 1280x1024 19" LG LCD.

First test (Intel PDF) led to a picture at the right of the screen.

Second (MiSTer FPGA) was a little more usable.

Here is the values i got so far (NOT WORKING, TEST AT YOUR OWN RISK), and note the sysclock adjustement needed:

```
// base from src/common/pico_scanvideo/vga_modes.c
/*
from https://www.intel.com/content/dam/support/us/en/programmable/support-resources/fpga-wiki/asset03/basic-vga-controller-design-example.pdf
Pixel Clock Frequency: 33.33MHz (33MHz will suffice)
Table 1: LCD timing parameters for the NEEK board
Horizontal Scanline Part Pixels Vertical Frame part Lines
Visible area                800 Visible area          480
Front Porch                 210 Front Porch            22
Sync Pulse                   30 Sync Pulse             13
Back Porch                   16 Back Porch             10
Whole line                 1056 Whole frame           525
*/
const scanvideo_timing_t vga_timing_wide_480_intel =
{
    // cf. https://misterfpga.org/viewtopic.php?t=2552
    // .clock_freq = 33000000,
    // 34.483 kHz
    .clock_freq = 34000000,

    .h_active = 800,
    .v_active = 480,

    // .h_front_porch = 210,
    .h_front_porch = 40,
    // .h_pulse = 30,
    .h_pulse = 48,
    // .h_total = 1056,
    .h_total = 928,
    .h_sync_polarity = 0,

    // .v_front_porch = 22,
    .v_front_porch = 13,
    // .v_pulse = 13,
    .v_pulse = 3,
    .v_total = 525,
    .v_sync_polarity = 0,

    .enable_clock = 1,
    .clock_polarity = 0,

    .enable_den = 1
};

const scanvideo_mode_t vga_mode_tft_400x240_intel =
{
    .default_timing = &vga_timing_wide_480_intel,
    .pio_program = &video_24mhz_composable,
    // .width = 400,
    // .height = 240,
    .width = 800,
    .height = 480,
    .xscale = 2,
    .yscale = 2,
};

/** @brief 400x240@50Hz, 4bpp, 16 colors */
const vgaboard_t vgaboard_400x240x4bpp = {
    .scanvideo_mode = &vga_mode_tft_400x240_intel,
    .freq_hz = 255,
    .depth = 4,
    .palette = ((uint16_t *)(&vgaboard_default_palette_4bpp)),
    // .sys_clock_khz = 33000L * 8L,
    .sys_clock_khz = 34000L * 8L,
};
```

kilograham, 2022-11-11:
Note that if you are driving a DPI display, then the timings are actually probably pretty flexible (it's not clear that a monitor will ever like an 800x480 mode)

CHiPs44, 2022-11-11:
My bad, I didn't know that TFT screens had those king of specific timings.

Even if 800x480 is not standard for VGA / VESA (cf. http://tinyvga.com/vga-timing for example), I thought this was not so weird and applicable to LCD or even CRT monitors, as the range is pretty wide.

closed 2022-12-26 by barbiani