raspberrypi/pico-playground #32: TURBO_BOOST breaks vga color

barbiani, 2022-11-18:
Hi,

I have noticed that enabling the TURBO_BOOST define breaks the correct colors display on the flash_stream sample vga_mode_640x480_60 .

Without TURBO_BOOST, the number is crisp and the blue gradient is perfect.

With TURBO_BOOST, a blue gradient turns to like 64 color dithered green.

Also, white blobs are leaking red pixels to the right.

Are these artifacts usually caused by vga timing error?

Thank you.

[image]

sfranzyshen, 2022-11-18:
just a guess but looks like it's related to set_sys_clock_khz() ... see other examples below 

https://github.com/raspberrypi/pico-playground/blob/e551a2d7606a468b00299e408ab44e62947926f9/scanvideo/mandelbrot/mandelbrot.c#L277
https://github.com/raspberrypi/pico-playground/blob/f0ce80c9d282b45fd1a3f882d92bce17bea4826d/scanvideo/sprite_demo/sprite_demo.c#L174

barbiani, 2022-11-18:
Looks like `set_sys_clock_khz(300000, true);` fixes the flash stream demo.

closed 2022-12-26 by barbiani