raspberrypi/pico-playground #34: Flash stream example producing messed color results

sandric, 2022-12-01:
I'm new to pico and failing to understand why I'm getting messed up image rendering with `flash_stream` example in scanline directory. Steps to reproduce screenshots I get:

1. run `./pack.sh` in `img` directory - it will produce 3 bin files and `cat` them into one `pack.bin`
2. run `sudo picotool load pack.bin -o 1003c000` to upload `pack.bin` into pico. The original command uses `uf2conv -f pico -b 0x1003c000 pack.bin -o pack.uf2` - but I don't know what `uf2conv` is and where to find it, so as this person on raspberry forum, and I took his suggested command with picotool - [uf2conv thread](https://forums.raspberrypi.com/viewtopic.php?t=305314)
3. build `flash_stream.uf2` and drag it into pico folder with file manager

Having done those I see such strange result, I know little about image processing and what exactly wrong with those:
[image]
[image]

I tried this on two different monitors, one big 24 inch monitor, and one small 7 inch 40 pin ttl monitor through the scaler - the result is equal.

Running `demo1` example with floating raspberry on a blue background works without any issue, so I doubt its a wiring problem.

I also tried to change packing command a bit - e.g. changing format and removing `d` in `./packtiles -sdf bgar5515 Voss_by_fortuneblues.png voss.bin` , it changes some things but colors are still messed.

barbiani, 2022-12-07:
Have a look at #32 

Also be sure that your image file has the exact resolution.

sandric, 2022-12-08:
@barbiani I tried applying this pr after I saw it - it didn't change anything. As for resolution of images, I have 2 screens as I said - one is small 7'', and I believe its maximum resolution is 1024x600, which is not exist as replacement for `#define VGA_MODE vga_mode_640x480_60` line. There is closer `vga_mode_tft_800x480_50` mode - but I got the same compilation issue as described [here](https://github.com/raspberrypi/pico-playground/issues/30). I tried to prevail the same aspect ratio by making images 640x375 pixels - but they just appeared cut, but same bug with wrong colors prevailed.

So I'm not sure what should I do at this point.

barbiani, 2022-12-08:
Make the image size exactly 640x480.

sandric, 2022-12-08:
@barbiani all images in `flash_stream/img` folder already 640x480 - https://github.com/raspberrypi/pico-playground/blob/master/scanvideo/flash_stream/img/Lighthouse_at_sunrise_by_Frenchie_Smalls.png

sandric, 2022-12-14:
Ok, I finally found the error - I simply connected all DAC resistors upside down, so that 500 Ohm was gpio0, etc.. Sorry, issue resolved.