raspberrypi/pico-playground #36: How to add usb printf to flash-stream dma example?

sandric, 2022-12-31:
I want to use draw 640x320 display while also monitoring capacitive touch panel every ~5ms via i2c, so I tried to elaborate on `flash-stream` example, but failed just to print out smth through usb. 

I tried adding second core entry to output usb string with 1 second delay, or vice versa - to move vga routine to second core, while outputting string in main one: in first case its just halts, in second while moving vga to second core - it manages to draw pictures normally but usb tty device not even recognized.

After debugging a bit - I found out that everything works just fine right until `__no_inline_not_in_flash_func(flash_bulk_read)` function is reached, and its first line `ssi_hw->ssienr = 0;` breaks it. I'm a newbie to pico and not familiar with this code.

So does anyone knows if its possible to use usb cdc for communication in this large pictures dma example and how to do it, or it is because pico uses all its resources to render images and have no compute power for usb stack (and potentially my i2c touch panel routine on top of it too)? Thx.

sandric, 2023-01-01:
I also not sure how to use even UART though - since vga uses all pins with uart buses using gpio0-gpio17, here's my pinout I took from pico documentation:

[image]

Can someone share schematic to use UART communication with vga 5-5-5 DAC?

lurch, 2023-01-01:
> Can someone share schematic to use UART communication with vga 5-5-5 DAC?

See section 3.3.1. in https://datasheets.raspberrypi.com/rp2040/hardware-design-with-rp2040.pdf

lurch, 2023-01-02:
I've personally not run any of these VGA examples, so can't offer help with those, but... 

>  or pico power is just not enough for anything except rendering 640x480 image

Raspberry Pi Pico is powerful enough to [run doom](https://rptl.io/doom-on-pico) :grinning: