raspberrypi/pico-playground #42: Issue with usb sound card

alessandromrc, 2023-05-24:
Hi, I've tried to implement 96KHz support but I get weird pitch and distorted audio apparently because there's some sample rate mismatch inside the audio system that creates this issue and I would like to know how to make it work with higher frequencies than 48k/16 bits.

Here's the code I changed:

```cpp
/*
 * Copyright (c) 2020 Raspberry Pi (Trading) Ltd.
 *
 * SPDX-License-Identifier: BSD-3-Clause
 */

#include <stdio.h>

#include "pico/stdlib.h"
#include "pico/usb_device.h"
#include "pico/audio.h"
#include "pico/audio_i2s.h"
#include "pico/multicore.h"
#include "lufa/AudioClassCommon.h"

// todo forget why this is using core 1 for sound: presumably not necessary
// todo noop when muted

CU_REGISTER_DEBUG_PINS(audio_timing)

// ---- select at most one ---
//CU_SELECT_DEBUG_PINS(audio_timing)

// todo make descriptor strings should probably belong to the configs
static char *descriptor_strings[] =
        {
                "PiAudio",
                "Test",
                "13124324"
        };

// todo fix these
#define VENDOR_ID   0x520au
#define PRODUCT_ID  0xf2ddu

#define AUDIO_OUT_ENDPOINT  0x01U
#define AUDIO_IN_ENDPOINT   0x82U

#undef AUDIO_SAMPLE_FREQ
#define AUDIO_SAMPLE_FREQ(frq) (uint8_t)(frq), (uint8_t)((frq >> 8)), (uint8_t)((frq >> 16))

#define AUDIO_MAX_PACKET_SIZE(freq) (uint8_t)(((freq + 999) / 1000) * 4)
#define FEATURE_MUTE_CONTROL 1u
#define FEATURE_VOLUME_CONTROL 2u

#define ENDPOINT_FREQ_CONTROL 1u

struct audio_device_config {
    struct usb_configuration_descriptor descriptor;
    struct usb_interface_descriptor ac_interface;
    struct __packed {
        USB_Audio_StdDescriptor_Interface_AC_t core;
        USB_Audio_StdDescriptor_InputTerminal_t input_terminal;
        USB_Audio_StdDescriptor_FeatureUnit_t feature_unit;
        USB_Audio_StdDescriptor_OutputTerminal_t output_terminal;
    } ac_audio;
    struct usb_interface_descriptor as_zero_interface;
    struct usb_interface_descriptor as_op_interface;
    struct __packed {
        USB_Audio_StdDescriptor_Interface_AS_t streaming;
        struct __packed {
            USB_Audio_StdDescriptor_Format_t core;
            USB_Audio_SampleFreq_t freqs[3];
        } format;
    } as_audio;
[... 557 lines cut ...]

    const struct audio_format *output_format;
    output_format = audio_i2s_setup(&audio_format_48k, &config);
    if (!output_format) {
        panic("PicoAudio: Unable to open audio device.\n");
    }

    ok = audio_i2s_connect_extra(producer_pool, false, 2, 96, NULL);
    assert(ok);
    usb_sound_card_init();

    multicore_launch_core1(core1_worker);
    printf("HAHA %04x %04x %04x %04x\n", MIN_VOLUME, DEFAULT_VOLUME, MAX_VOLUME, VOLUME_RESOLUTION);
    // MSD is irq driven
    while (1) __wfi();
}
```

In the CMake I did this change here:

```
AUDIO_FREQ_MAX=96000
```

Hope to get a response back as this project seems dead. I was thinking if it would also make more sense to recreate the entire soundcard using TinyUSB at this point...

alessandromrc, 2023-07-30:
Apparently the issue is related to USB buffers which I'm actually looking into right now and I've got 96kHz 16 bit working without any issue... will look forward at adding more sample rates and maybe higher up the bitrate to 24

AnkitBhasin10, 2025-07-14:
Hi @alessandromrc 

What changes you did to make it work at 96khz? I also recently tried it and am also getting distorted sound. I am implementing this for PS5 since it supports UAC1.0. Not sure if it supports 96khz output but will try.

Thanks

alessandromrc, 2025-07-28:
> Hi [@alessandromrc](https://github.com/alessandromrc)
> 
> What changes you did to make it work at 96khz? I also recently tried it and am also getting distorted sound. I am implementing this for PS5 since it supports UAC1.0. Not sure if it supports 96khz output but will try.
> 
> Thanks

Hey @AnkitBhasin10

I am sorry to let you know that I haven't had any time since then to thinker around the Pi Pico, I thought people already solved this issue...

The only thing I remember doing ages ago was changing some parameters within the USB Buffers on the TinyUSB library and the "distorted sound" disappeared.

Also, you could switch to a Teensy board if you are trying to get the best audio out of your money, obviously coupled together with a nice low S/N ratio DAC.