raspberrypi/pico-playground #50: Confusion re building examples in `pico-playground`

seamusdemora, 2024-09-20:
Sorry to have to ask, but I cannot get this to work: 

I need to use the `sleep` and/or `dormant` mode for an *off-grid* Pico project. I plan to use a [RV-3028](https://www.microcrystal.com/en/products/real-time-clock-rtc-modules/rv-3028-c7) as the RTC/interrupt source due to its accuracy and low power consumption. 

I planned on using the examples in `pico-playground` to get me started, but I've *"missed a beat"* somewhere along the line: 

   ```bash
   $ cd ~/pico/pico-playground
   $  mkdir build 
   $ cd build
   $ export PICO_SDK_PATH=../../pico-sdk              # don't know if this was needed or not 
   $ cmake .. 
   $ cd sleep/hello_sleep
   $ make -j2 
    . . .
   [100%] Built target hello_sleep 
   $ 
   ``` 

So here's where I'm confused. I had to descend two levels below `./build` - into `sleep/hello_sleep` to get things to work. This doesn't *feel right*... 

Should I be copying `hello_sleep` into `/home/pi/pico/examples/build`?? - instead of trying to set up `pico-playground` as a stand-alone set of examples??

TIA!

lurch, 2024-09-20:
> So here's where I'm confused. I had to descend two levels below `./build` - into `sleep/hello_sleep` to get things to work. This doesn't _feel right..._

IIRC this is exactly the same way that `pico-examples` works. Note that the examples in `pico-examples` need `pico-sdk` to be present (specified with `PICO_SDK_PATH=...`), but the examples in `pico-playground` _may_ also need `pico-extras` to be present (specified with `PICO_EXTRAS_PATH=...`)

> Should I be copying `hello_sleep` into `/home/pi/pico/examples/build`?? - instead of trying to set up `pico-playground` as a stand-alone set of examples??

You shouldn't be using the `CMakeLists.txt` from a pico-examples or pico-playground example for your _own_ projects, as these are all tightly integrated with the top-level `CMakeLists.txt` in those repos.

To create your own project (independent from the `pico-playground` repo), see the "Manually Create your own Project" section in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

> `$ export PICO_SDK_PATH=../../pico-sdk              # don't know if this was needed or not`

If you're jumping around between different build directories, it's probably safer to set this to an absolute path, rather than a relative path.

kilograham, 2024-09-20:
> So here's where I'm confused. I had to descend two levels below ./build - into sleep/hello_sleep to get things to work. This doesn't feel right...
> $ make -j2 

Make from the top-level builds everything in `pico-playground` by default

you could do

```
make -j2 hello_sleep
```

instead.

seamusdemora, 2024-09-23:
Following the instructions in `getting-started-with-pico-2024-08-21.pdf` for "Manually Create your own Project" does not seem to work with the `hello_sleep` or `hello_dormant` examples from pico-playground. I think I followed the instructions, but got an error when running `make`: 

```bash
$ pwd
~/pico
$ mkdir sleep_dormant
$ mkdir sleep_dormant/build
$ cd sleep_dormant
$ cp ../pico-sdk/external/pico_sdk_import.cmake .
$ cd build
$ export PICO_SDK_PATH=../../pico-sdk
$ cmake ..
...
-- Build files have been written to: /home/pi/pico/sleep_dormant/build
$ make 
...
[ 14%] Building C object CMakeFiles/sleep_dormant.dir/sleep_dormant.c.obj
/home/pi/pico/sleep_dormant/sleep_dormant.c:9:10: fatal error: pico/sleep.h: No such file or directory
    9 | #include "pico/sleep.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/sleep_dormant.dir/build.make:76: CMakeFiles/sleep_dormant.dir/sleep_dormant.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1748: CMakeFiles/sleep_dormant.dir/all] Error 2
make: *** [Makefile:91: all] Error 2 
---
$ find ~/pico -type f -iname "sleep.h"
/home/pi/pico/pico-sdk/lib/btstack/port/samv71-xplained-atwilc3000/ASF/sam/drivers/pmc/sleep.h
/home/pi/pico/pico-extras/src/rp2_common/pico_sleep/include/pico/sleep.h
$ 
``` 

Sorry, but I'm at a complete loss here. It's almost like `pico-playground` and `pico-extras` are completely cut off from the rest of the install??

lurch, 2024-09-23:
Looks like @kilograham has given you the fix for this in your other issue? https://github.com/raspberrypi/pico-feedback/issues/415

@nathan-contino Do you think it's worth documenting how to use `pico-extras` in your own project in the getting-stared documentation? :thinking:

seamusdemora, 2024-10-01:
> Looks like @kilograham has given you the fix for this in your other issue? [raspberrypi/pico-feedback#415](https://github.com/raspberrypi/pico-feedback/issues/415)
>
> @nathan-contino Do you think it's worth documenting how to use `pico-extras` in your own project in the getting-stared documentation? 🤔
> 
> EDIT: Ahhh, I've just seen that this **is** actually documented towards the bottom of https://github.com/raspberrypi/pico-extras/blob/master/README.md ! 

I got this working using the following line in `CMakeLists.txt`:

```
include(/home/pi/pico/pico-extras/external/pico_extras_import.cmake)
``` 

This was the only thing that seemed to work. Apparently I don't understand the instructions well at all: 

I checked : 
```bash
$ echo $PICO_EXTRAS_PATH
/home/pi/pico/pico-extras
``` 

But there seems to be no other `include` specification that works; e.g.
```
include(external/pico_extras_import.cmake)
```
did not work. ??