raspberrypi/pico-playground #52: popcorn doesnt work

Kr666you, 2024-10-19:
Hello. I have connected my sdcard like in https://datasheets.raspberrypi.com/rp2040/hardware-design-with-rp2040.pdf
But when i connect my pico to my monitor, I get a blue and black screen. I heard someone say that this is because of the "framebuffer overflowing" or something like that.
Flashed my sdcard with: sudo dd if=bunny.pl2 of=/dev/sdb bs=4M status=progress
bunny.pl2 file from https://github.com/raspberrypi/pico-playground/tree/master/apps/popcorn
No errors when flashing. I dont know what is happening

lurch, 2024-10-22:
Perhaps it's a wiring problem? Perhaps the 'popcorn' application is trying to use a display-mode that your monitor doesn't support?

"XXXX doesn't work, please help" is a question much more suited for https://forums.raspberrypi.com/viewforum.php?f=143
GitHub is for reporting and tracking specific bugs.

Kr666you, 2024-10-23:
> Perhaps it's a wiring problem? Perhaps the 'popcorn' application is trying to use a display-mode that your monitor doesn't support?
>
> "XXXX doesn't work, please help" is a question much more suited [...]

nope. rewired it 5 times. 
its not a display-mode problem. if it was then probably it wouldn't even display anything.
i tried compiling without -DPICO_BOARD and with -DPICO_BOARD=vgaboard.
also tried putting my bunny.pl2 file onto a new partition (gpt) with the same name as the file but nothing